RPi-Distro/raspi-config #138: GPIO fan does not work

chaptergy, 2020-10-29:
I have noticed a new GPIO Fan option in performance options and decided to try it. However it does not seem to work. 

I have connected my fan to the pins, black to physical pin 6, and red to physical pin 8, so GPIO 14. Then I went into `raspi-config` > 'Performance Options' > 'Fan'. 'Enable fan temperature control' > 'Yes'. 'Which GPIO?' > 14. 'At what temperature?' > 60. Then I rebooted my Pi. However even after heating up the Pi with a hairdryer and `cat /sys/class/thermal/thermal_zone0/temp` reporting values above 60000, the fan did not turn on.

I am on a Raspberry Pi 4 and the fan a generic pi-fan.

XECDesign, 2020-10-29:
I'm not familiar with this option, but you don't also have serial or anything like that enabled on that pin? What does `raspi-gpio get 14` say?

Maybe @spl237 will know more about it.

spl237, 2020-10-29:
I think it will only work with the most recent firmware, kernel etc - maybe you haven't updated to those? Also, I believe you need a three-wire fan, not a two wire - so there should be GND and 5V connections, and the specified pin connects to a control wire on the fan.

chaptergy, 2020-10-29:
Everything should be up to date, it is a freshly set up Pi with `apt upgrade` run. But if it needs a three-wire fan it would explain the behaviour, I assumed it would just turn the pin on or off, most three-pin-fans have a more fine grained speed control.

maxscience, 2020-11-03:
Following. I also have a two wired fan and a Raspi4 with latest OS and updates. So there's no way of using temperature control? Right now do you also let the fan run all the time @chaptergy ?

Void-Frost, 2020-11-03:
I also am curious how you are supposed to used the GPIO fan, and I tried it with a stress test and it just didn't kick on.
Looking forward to any possible answers, and I will be experimenting for sure to see if I can get it to work.

nashneshev, 2020-11-07:
I was trying to make this work as well and I believe I figure out the problem.
First, the way this functionality is supposed to work is that the selected pin output '1' (3.3V) when the SoC reaches the specified temperature and then goes back to '0' (0V) when the temperature drops 10 degrees lower. I wouldn't attach a fan directly to the GPIO pin - the current output should not be enough to power a fan. Instead it should drive a transistor that then turns the fan on or off. So you need a couple of extra elements.

Still, there is a bug in raspi-config. According to the gpio-fan overlap description at 
 https://github.com/raspberrypi/firmware/tree/master/boot/overlays the parameters are `gpiopin` and `temp`, while raspi-config is setting `gpio_pin` and `temp`. The pin parameter is invalid and from there the temp is ignored as well. 

@chaptergy, to make this work you can try editing your /boot/config.txt and remove the extra underscore added by raspi-config, so from:
`dtoverlay=gpio-fan,gpio_pin=14,temp=60000`
to
`dtoverlay=gpio-fan,gpiopin=14,temp=60000`
Then you will see a logical '1' on the pin when the Pi goes over 60 degrees and logical 0 when it gets under 50 degrees. Again, I wouldn't connect a fan directly to the pins, but would use the pin to control the fan.

Repo maintainers, would you like a PR with the proposed fix to raspi-config?

ghollingworth, 2020-11-08:
Yes, sorry it should be gpiopin, my bad for not checking properly!

But as mentioned above, you cannot use this to turn on and off a two wire fan without external electronics...  This is for three or four wire fans which have a PWM input (the gpio-fan overlay doesn't drive PWM, it just powers the fan on and off at the determined temperature)...

Also note I usually set this to 80 degrees, since otherwise it'll be noisy, but there is no extra value in being able to run the chip at a lower temperature anyway...  This does mean that with all the power consumption improvements since we launched Pi 4 it's actually quite difficult to get there, but compiling the kernel with -j4 or running cpuburn will get you there!

Gordon

nashneshev, 2020-11-08:
@ghollingworth, are there any articles / recommendations on what's a healthy temperature range for the Pi? Some of my friends worry when their Pies get over 55 deg C, so I added an option in my fix to set the temperature lower than 60. Also, the gpio-fan overlay's default temperature is 55, same seems to be for the rpi-poe. Why were these temperatures chosen? Do they provide longer life to the board?

ghollingworth, 2020-11-08:
The silicon is tested and will work fine up to around 120 degrees, the real temperature sensitivity is in the capacitors.  But anything less than 85 degrees is fine...

The only reason we set this maximum is for safety, even at 85 degrees it gets really rather hot to the touch!

80 degrees max is fine

spl237, 2020-11-08:
I've pushed a fix for the typo in the name of the gpiopin parameter - should be live in apt within a day or so.

nashneshev, 2020-11-08:
Thank you for the quick turnaround!

spl237, 2020-11-08:
No problem - many thanks for reporting the issue, and apologies that I pushed a fix before I spotted your PR!

closed 2020-11-08 by spl237

maxscience, 2020-11-19:
> But as mentioned above, you cannot use this to turn on and off a two wire fan without external electronics... This is for three or four wire fans which have a PWM input (the gpio-fan overlay doesn't drive PWM, it just powers the fan on and off at the determined temperature)...
> 

@ghollingworth why so?
I guess the goal of this feature would be to just simply remove power to the pin where the 2wire fan is connected... Then enable it when temperature goes above threshold. Many raspi fans sold with cases are just 2wires.
@nashneshev does it work properly for you now with the 2wire fan?
Thank you

ghollingworth, 2020-11-19:
You are suggesting powering the fan from the GPIO pin itself.  The fans generally don't run from 3v3 (they tend to run from 5v), plus there is a limit on the current from the GPIO pin which wouldn't be enough to power the fan.  In fact it'd just damage the GPIO pin...  The maximum rating for a single GPIO pin is 18mA but even a low strength fan needs 60mA to start.

maxscience, 2020-11-19:
@ghollingworth I'm not suggesting actually, this is how my raspi fan works. It's designed to be powered by the gpio pins and there are so many raspi4 cases with fans like that.