RPi-Distro/raspi-config #141: Controlling VC I2C with raspi-config?

hailfinger, 2020-11-22:
Hi,

it seems that dtparam=i2c_vc=on in /boot/config.txt can't be set with raspi-config. Would you accept a patch / pull request adding that functionality? I'm using i2c_vc on a Pi 4 to determine which camera or HDMI-to-CSI capture card is present, and having an option in raspi-config would make sure I won't have to roll my own config editing utility.
Background: I'm trying to build a universal streaming source image (based on Raspberry Pi OS) which should be able to handle all Pi cameras, HDMI-to-CSI capture cards, USB CVBS capture cards, RTMP streams and more. Having autodetection of the available video and audio sources is essential to usability in this case.

XECDesign, 2020-11-23:
I think the idea is not to encourage users to use i2c_vc, as they may run into problems with some configurations. It has become less of an issue over the past few years, but I'd still avoid making it easy for a beginner to shoot themselves in the foot.

hailfinger, 2020-11-23:
Right now i2c_vc seems the only feasible way to autodetect what's present on the CSI interface. Should there be another (supported) method, I'm all ears. Right now is seems this script https://gist.github.com/Hermann-SW/3c0387c4340e10866af32cc5a3d21036 or variants are a reasonably reliable CSI detection. vcgencmd get_camera unfortunately does not play nice with the tc358743 dtoverlay.

XECDesign, 2020-11-24:
That is more @6by9's area.

If using i2c_vc is the only thing that works, I'm not saying advanced users shouldn't use it, only that it's a bit outside of what raspi-config is for.

6by9, 2020-11-24:
XECDesign is correct - i2c_vc is still regarded as a non-standard option. If you're using ```dtoverlay=tc358743``` then it's not required anyway as that overlay enables the I2C bus (as it should), as do all the other sensor overlays.
Use of i2c_vc when either the camera or DSI display are being controlled by the firmware risks issues.

You may be on a hiding to nothing as there is no obligation to load the i2c-dev module to provide userspace access to the I2C buses.

If looking for information as to what sensor(s) the firmware has detected, then use the vc.camera_info component to enumerate the camera names and resolutions, same as raspistill does - https://github.com/raspberrypi/userland/blob/master/host_applications/linux/apps/raspicam/RaspiHelpers.c#L90

If using the kernel drivers via a dtoverlay, then use the media controller API to find the device name - the programatical equivalent to ```media-ctl -p -d /dev/media2```.

imx290, imx327, and imx477 are all on I2C address 0x1a.
ov9281 and ov7251 both use I2C address 0x60.
Solely using the I2C address does not uniquely identify a sensor.

hailfinger, 2020-11-26:
Thanks @6by9 , that is very helpful info.
My problem is that I want to autodetect the hardware configuration which is one of three options:
- a Raspberry Pi HQ Camera
- a HDMI-to-CSI bridge
-no CSI device attached

If the tc358743 dtoverlay is present, AFAICT the Raspberry Pi firmware won't detect a Pi HQ Camera. Your hint about i2c_vc being activated by the tc358743 dtoverlay is really helpful as well.
If the overlay is not present, it seems that the code you pointed to will tell me tell me if a tc358743 bridge or any of the firmware-supported cameras is present.

I'll try to come up with code according to your guidelines and will report back.

6by9, 2020-11-26:
If DT has an enabled csi0 or csi1 node, then the firmware will refuse to enable that port, and in turn won't detect a camera module connected to it. That is important if you use ```dtoverlay=imx477``` for example as the HQ camera driver for use with libcamera, so we don't want interference from the firmware.

My suggestion would be to:
- first try creating a MMAL vc.camera_info component. If that reports anything then job done.
- Otherwise probe via media controller. If you're wanting a potential template for that, then look at the libcamera code.
- If media controller reports nothing, then you have no device connected (or someone's done something funny in their kernel config).

closed 2020-12-11 by no-response[bot]