RPi-Distro/raspi-config #142: No network option available inside raspi-config after running apt upgrade

GChuf, 2020-11-26:
I'm running a minimal raspberry pi OS.
I've noticed the error appeared after running `apt update && upgrade`. I've then tried to run only `apt install raspi-config` on a fresh install and the results were the same.

Running kernel 5.4.72 after upgrade, or 5.4.51 when just upgrading raspi-config on a `Pi 3 Model B rev1.2`.

[image]

XECDesign, 2020-11-26:
Doesn't seem to be an error, just a re-organised menu.

GChuf, 2020-11-26:
Where are the network options then @XECDesign ?
And why is number 7 missing?

XECDesign, 2020-11-26:
Which option are you after?

GChuf, 2020-11-28:
I was looking for wifi setup in particular. I think I saw some other network options somewhere ...

I was able to connect to wifi by manually editing wpa_supplicant.conf file though.

XECDesign, 2020-12-01:
That seems to be under System Options now (1>S1)

closed 2020-12-01 by XECDesign