RPi-Distro/raspi-config #147: done everything still not getting sound through 3.5 mm jack

Nikhildhabu, 2020-12-22:
im using VGA monitor and 3.5mm audio out , usb based audio in. 
I have done rasp.conf > audio > headphones
sudo amixer cset numid=3 1

XECDesign, 2021-01-04:
Are you running the lite or desktop image?

@spl237 any idea?

spl237, 2021-01-04:
sudo amixer cset numid=3 1 doesn't work any more! That's the old way of setting the output before we changed to separate ALSA devices, and long before we changed to Pulseaudio.

Without knowing what version of what image is being used, it's hard to say exactly what the problem is, but sudo amixer cset isn't the right way of doing things any more...

Blackbox-git, 2021-01-19:
I have the same problem with my Pi. I am using a Raspberry Pi 2 Model B Rev 1.1 and the Raspberry Pi OS from 11.01.2021 with all Updates.

```processor       : 0
model name      : ARMv7 Processor rev 5 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpa                                                                                                                                                                                                                                   e evtstrm
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xc07
CPU revision    : 5

processor       : 1
model name      : ARMv7 Processor rev 5 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpa                                                                                                                                                                                                                                   e evtstrm
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xc07
CPU revision    : 5

processor       : 2
model name      : ARMv7 Processor rev 5 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpa                                                                                                                                                                                                                                   e evtstrm
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xc07
CPU revision    : 5

processor       : 3
model name      : ARMv7 Processor rev 5 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpa                                                                                                                                                                                                                                   e evtstrm
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xc07
CPU revision    : 5

Hardware        : BCM2835
Revision        : a01041
Serial          : 000000003b884749
Model           : Raspberry Pi 2 Model B Rev 1.1
```
The image is output via the HDMI port. The sound is to be output via the headphone connection.

A configuration in raspi-config does not create the file "asound.conf". Raspi-Config gives me as selection 0 HDMI 1 and 1Headphone. Of course I selected Headphone. 

After I save the file manually in the directory /home/pi/asound.conf, it is deleted at a reboot.

In both cases the sound output is not done via the headphone port.

spl237, 2021-01-19:
To repeat XECDesign's question above - are you using a desktop image or a lite image?

Blackbox-git, 2021-01-19:
Sorry, i use Raspberry Pi OS with desktop.

spl237, 2021-01-19:
OK, I've just tried the 11/1/21 image on a Pi 2 here, and audio switching works correctly with both the audio switcher on the taskbar and with raspi-config.

Because the system now uses PulseAudio, asound.conf is no longer used, and it will be deleted because some applications use it in preference to the device selected by PulseAudio.

Which application are you using to play sound? It is possible that whichever application you are using does not work properly with Pulseaudio - Sonic Pi, for example, bypasses PulseAudio and only sends output to HDMI. If you play a file with the command line tool aplay, it should correctly route to the device selected in PulseAudio.

Blackbox-git, 2021-01-19:
I use aplay for playback. This I call without further options with the name of the file to play. I have no mouse or keyboard and therefore use raspi-config for the setting.

I can not see on the graphical interface what interface is set. However, I do not understand why it works for you and not for me. Are there commands in which I change the sound output in Pulse Audio via Console?

spl237, 2021-01-19:
"pactl list short sinks" lists the valid audio output devices. 

There is an index number at the start of each line. Index 0 should be alsa_output.platform-bcm2835_audio.digital-stereo, which is the HDMI port. Index 1 should be alsa_output.platform-bcm2835_audio.analog-stereo, which is the HDMI port. To set the output, use the command "pactl set-default-sink" followed by the appropriate index, so:

"pactl set-default-sink 0" should set output to HDMI
"pactl set-default-sink 1" should set output to the headphones

closed 2021-01-20 by no-response[bot]