RPi-Distro/raspi-config #150: Locale is not set after generating them

MPeti1, 2021-01-29:
I just now discovered that `locale` returns everything set as POSIX, instead of the locale I selected in raspi-config's appropriate menu.

[image]

I'm not using the default user account, I've followed "Changing your username" [here](https://www.raspberrypi.org/documentation/configuration/security.md), including the deletion of the default account.

I was able to fix it in a few hours time by including export statements for LANGUAGE, LANG, and LC_TIME (because I rather want to use my country's date format than the US one) in ~/.bashrc, but I'm not sure this is the proper way of doing that.
Sadly I haven't found out in that time why that happens, so I came here to report a potential bug.

From what I've read it might be important to note that I almost always log in through SSH, but by running `env` I haven't seen any locale vars to be set up until I have done it in ~/.bashrc.

XECDesign, 2021-02-04:
Is this after a reboot or straight after running raspi-config?

MPeti1, 2021-02-04:
It was after a reboot. Actually more than one, because I didn't notice for some time that it wasn't set up correctly.

garyk10, 2021-09-20:
I found a typo in raspi-config that is causing this.  It is in the do_change_locale() function in /usr/bin/raspi-config.  Line 495 has an extra space in it.  **It should change from this:**
if ! LOCALE_LINE="$(grep "^$LOCALE " /usr/share/i18n/SUPPORTED)"; then

**to this:**
if ! LOCALE_LINE="$(grep "^$LOCALE" /usr/share/i18n/SUPPORTED)"; then

XECDesign, 2021-10-04:
OP was using raspi-config through the menu. The path on line 495 you reference is only taken in non-interactive mode.

garyk10, 2021-10-06:
Alright, that's true.  I thought I had reproduced the OP's issue, but I hadn't.

To OP: check your /etc/default/locale to see what's in there

garyk10, 2021-10-12:
If both en_US.UTF-8 and your [my country].UTF-8 locales have both been generated, then you should be okay.  Or maybe line endings or other whitespace are a problem.  Regardless, I suggest rerunning this command, which should correct your /etc/default/locale file:
`sudo dpkg-reconfigure locales`

closed 2021-10-19 by MPeti1