RPi-Distro/raspi-config #155: Selecting Display > Resolution does not work - returns to the main menu

dankcushions, 2021-05-03:
I am unable to select the Resolution menu as it always quits back to the main menu immediately. This is via Raspi OS Lite 5.10.11-v7l+.

Potentially relevant parts of `config.txt`:
```
# uncomment to force a specific HDMI mode (this will force VGA)
hdmi_group=2
hdmi_mode=82
```
(i have to manually select HDMI modes otherwise it defaults to 4k which has bad performance - if I remove these lines it still fails)
```
dtoverlay=vc4-fkms-v3d
max_framebuffers=2

[all]
overscan_scale=1
```

wagnerch, 2021-12-20:
Same issue here with kms on Bullseye.  It appears when mutter is used as a WM it has it's own configuration to manage monitors & resolutions.  Apparently GNOME Settings knows how to configure it.  Basically lightdm is running the display setup script written by arandr (screen configuration) but mutter is changing it immediately after.

I managed to get a persistent resolution saved by creating this Python script:
```python
#!/usr/bin/env python3
import sys
from gi.repository import Gio, GLib

bus = Gio.bus_get_sync(Gio.BusType.SESSION, None)
proxy = Gio.DBusProxy.new_sync(
    bus,
    Gio.DBusProxyFlags.NONE,
    None,
    "org.gnome.Mutter.DisplayConfig",
    "/org/gnome/Mutter/DisplayConfig",
    "org.gnome.Mutter.DisplayConfig",
    None
)

resources = proxy.call_sync(
    "org.gnome.Mutter.DisplayConfig.GetCurrentState",
    None,
    Gio.DBusCallFlags.NONE,
    1000,
    None
)

resources = resources.unpack()
serial = resources[0]
monitors = resources[1]
#print(resources)
print("serial:", serial)
print("monitors:")
for monitor in monitors:
  print(monitor[0][0:3])
  for mode in monitor[1]:
    print("  ", mode)

if len(sys.argv) != 4:
  sys.exit(1)

resolution = GLib.Variant("(uua(iiduba(ssa{sv}))a{sv})", (
    serial,
    2, # 0=verify, 1=temp, 2=persistent
    [ ( 0, 0, float(sys.argv[3]), 0, True, [ ( sys.argv[1], sys.argv[2], [] ) ] ) ],
    []
  )
)

result = proxy.call_sync(
    "org.gnome.Mutter.DisplayConfig.ApplyMonitorsConfig",
    resolution,
    Gio.DBusCallFlags.NONE,
    1000,
    None
)

print(result)
```

Usage:
`./mutter-set-display-resolution`

Should display monitors and modelines:
```
serial: 2
monitors:
('HDMI-1', 'ACR', 'ED347CKR')
   ('3440x1440@59.998870849609375', 3440, 1440, 59.998870849609375, 1.0, [1.0, 2
.0, 3.076923131942749], {'is-preferred': True})
   ('3440x1440@99.992500305175781', 3440, 1440, 99.99250030517578, 1.0, [1.0, 2.
0, 3.076923131942749], {})
   ('3440x1440@99.986869812011719', 3440, 1440, 99.98686981201172, 1.0, [1.0, 2.
0, 3.076923131942749], {})
   ('2560x1440@59.950550079345703', 2560, 1440, 59.9505500793457, 1.0, [1.0, 2.0, 3.076923131942749], {})
   ('2560x1080@59.994792938232422', 2560, 1080, 59.99479293823242, 1.0, [1.0, 2.0, 3.076923131942749], {'is-current': True})
...
```

The second run to set the mode is:
`./mutter-set-display-resolution HDMI-1 2560x1080@59.994792938232422 1.0

1st arg: connector
2nd arg: mode line
3rd arg: scale (1.0)

The script will call the Mutter display config service over DBus and set a persistent configuration.

BenMcLean, 2022-02-26:
Is there anything we need to do to get this crippling problem some attention?

XECDesign, 2022-02-28:
@spl237

spl237, 2022-02-28:
The resolution menu in raspi-config is only usable with legacy mode video drivers; with the KMS driver which is now standard, the command-line xrandr tool should be used instead. This option shouldn't be shown in the raspi-config menu any more; I will remove it.

spl237, 2022-02-28:
Also - you should not be using the line dtoverlay=vc4-fkms-v3d on any of our images any more - that loads the fake KMS driver, which is no longer supported. That line should either be ...vc4-kms-v3d, or it should be absent altogether to access the legacy video driver, but that is not an option on Pi 4 or Pi 400, which have to use KMS. What you have in your config.txt looks unsupported to me.

lurch, 2022-02-28:
> you should not be using the line dtoverlay=vc4-fkms-v3d on any of our images any more - that loads the fake KMS driver, which is no longer supported.

Oh! I didn't know that.
Does that means that [this](https://github.com/raspberrypi/documentation/blob/develop/documentation/asciidoc/computers/configuration/raspi-config.adoc#gl-driver) and [this](https://github.com/raspberrypi/documentation/blob/develop/documentation/asciidoc/accessories/camera/libcamera_software.adoc#select-the-correct-graphics-driver) needs updating Simon? (ping also @aallan as I guess this also impacts https://github.com/raspberrypi/documentation/issues/2310 )

spl237, 2022-02-28:
Err - pass. All I know is that I have modified raspi-config so that the only GL driver options are KMS on Pi 4 / 400, and KMS or legacy on older Pis. fKMS is not available in raspi-config any more, so my understanding is that people should not be using it; it still works (presumably...), but I don't think future support for it is guaranteed. Worth checking the official position with GH or similar rather than taking my word for it though...

spl237, 2022-02-28:
But yes, at a quick look, those pages are definitely out of date - the KMS driver is now the default and is no longer experimental, for a start.

aallan, 2022-02-28:
I guess this ties into raspberrypi/documentation#2310 and also raspberrypi/documentation#2061

XECDesign, 2022-03-09:
This is looking like a wontfix. I don't think there's going to be effort made to support resolution setting for legacy or FKMS. If there's a way to get raspi-config to adjust resolution on the fly with KMS that works with or without X, that would be great, but sounds like a separate issue.

closed 2022-03-09 by XECDesign

dankcushions, 2022-03-09:
Fair, though rather than just exit out and do/say nothing, it would be better in the scenario of KMS + outside of X that the option adjusted config.txt, but prompted for a reboot.

BenMcLean, 2022-03-09:
> Fair, though rather than just exit out and do/say nothing, it would be better in the scenario of KMS + outside of X that the option adjusted config.txt, but prompted for a reboot.

Since the active devs want to not spend time on this (totally understandable) I brought up the idea of just removing the Resolution menu altogether [over here](https://github.com/raspberrypi/Raspberry-Pi-OS-64bit/issues/200#issuecomment-1063239679).

spl237, 2022-03-09:
The option has already been removed for the next release.