RPi-Distro/raspi-config #159: Disable the new GL driver on Pi 4, now see error on boot?

dachshund-digital, 2021-05-25:
Disable the new GL driver on Pi 4, now see error on boot?

`# systemctl status gldriver-test
● gldriver-test.service - Check for v3d driver
   Loaded: loaded (/lib/systemd/system/gldriver-test.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Tue 2021-05-25 00:33:41 UTC; 6min ago
 Main PID: 416 (code=exited, status=1/FAILURE)

May 25 00:33:41 localhost systemd[1]: Starting Check for v3d driver...
May 25 00:33:41 localhost gldriver_test.sh[416]: /usr/lib/systemd/scripts/gldriver_test.sh: line 12: /usr/share/X11/xorg.conf.d/99-fbturbo.conf: No such file or directory
May 25 00:33:41 localhost systemd[1]: gldriver-test.service: Main process exited, code=exited, status=1/FAILURE
May 25 00:33:41 localhost systemd[1]: gldriver-test.service: Failed with result 'exit-code'.
May 25 00:33:41 localhost systemd[1]: Failed to start Check for v3d driver.
`

Raspberry Pi OS lite installed on Pi4.  The GL driver is applicable only to X11 environment right?  This error should not be occurring thus?

XECDesign, 2021-05-26:
@spl237 The `[Unit]` section of `gldriver-test`'s `gldriver-test.service` should probably have `ConditionPathIsDirectory=/usr/share/X11/xorg.conf.d`?

dachshund-digital, 2021-05-26:
I will check,.. Nope, or I should say, yeah, maybe that is the fix, the unit file is pretty basic...

`[Unit]
Description=Check for v3d driver

[Service]
Type=oneshot
ExecStart=/usr/lib/systemd/scripts/gldriver_test.sh
RemainAfterExit=yes

[Install]
WantedBy=multi-user.target
`

Since I am running lite, none of the is applicable per se.  Seems lite something that was missed when the lite image was established.  For now I disabled the service to avoid the error, which as far as I can see, just informational if you are not using the GL driver in the first place.

The actuall script executed...

`#!/bin/bash

if grep -s -q okay /proc/device-tree/soc/v3d@7ec00000/status \
                   /proc/device-tree/soc/firmwarekms@7e600000/status \
                   /proc/device-tree/v3dbus/v3d@7ec04000/status ||
   [ "$(arch)" == aarch64 ]; then
        if [ -e /usr/share/X11/xorg.conf.d/99-fbturbo.conf ] ; then
                rm /usr/share/X11/xorg.conf.d/99-fbturbo.conf
        fi
else
        if ! [ -e /usr/share/X11/xorg.conf.d/99-fbturbo.conf ] ; then
                cat > /usr/share/X11/xorg.conf.d/99-fbturbo.conf << EOF
# This is a minimal sample config file, which can be copied to
# /etc/X11/xorg.conf in order to make the Xorg server pick up
# and load xf86-video-fbturbo driver installed in the system.
#
# When troubleshooting, check /var/log/Xorg.0.log for the debugging
# output and error messages.
#
# Run "man fbturbo" to get additional information about the extra
# configuration options for tuning the driver.

Section "Device"
        Identifier      "Allwinner A10/A13 FBDEV"
        Driver          "fbturbo"
        Option          "fbdev" "/dev/fb0"

        Option          "SwapbuffersWait" "true"
EndSection
EOF
        fi
fi
`

Running the script it fails with...

'/usr/lib/systemd/scripts/gldriver_test.sh: line 12: /usr/share/X11/xorg.conf.d/99-fbturbo.conf: No such file or directory'

For Pi OS lite image.