RPi-Distro/raspi-config #166: Can't disable predictable network interface names

count0x30, 2021-09-15:
Upgrading Raspbian to Bullseye enabled predictable network interface names on my system, and raspi-config isn't working to turn them off.
https://github.com/RPi-Distro/raspi-config/blob/64d5c2c25d67002bdc6b9dc891cc92e1f27f7831/raspi-config#L1906-L1912
Shouldn't `net.ifnames=0` be written back to /boot/cmdline.txt when disabling predictable network i/f names?

XECDesign, 2021-09-22:
This seems to be the culprit: https://salsa.debian.org/systemd-team/systemd/-/commit/dcd337b646ae301a6f8245afabc69f8d1ddc3b34

I'll update raspi-config to override that `.link` as well.

Thank you

closed 2021-09-23 by XECDesign

count0x30, 2021-09-29:
Just tested it out on my Pi 0 with a USB to Ethernet adapter, and I could enable and disable predictable network interface names through raspi-config.  Thanks!

mschlenstedt, 2022-12-21:
I have the same problem here - upgrading from buster to bullseye enabled predictable network interface names on my system (tested on Pi3 and Pi1).

Adding  net.ifnames=0 to /boot/cmdline.txt solved the problem.

I think you missed adding this option (it is removed when $RET=0, but not added when $RET=1

https://github.com/RPi-Distro/raspi-config/blob/0fc1f9552fc99332d57e3b6df20c64576466913a/raspi-config#L2155