RPi-Distro/raspi-config #169: Feature request in raspi-config, interface options - add disable WiFi / Bluetooth options.

CaptainMidnight, 2021-11-14:
Just an observation and curiosity, as raspi-config is a user friendly way of making /boot/config.txt and other system file configuration changes, it appears strange that both WiFi and Bluetooth are not listed in the "Interface Options" section?

i.e it contains no option of being to disable WiFi or Bluetooth functionality using this configuration utility.

Yes fully understand this can be achieved manually by editing /boot/config.txt, but as a user friendly configuration utility, it seems odd that it isn't already included in the interface options.

CaptainMidnight, 2021-11-14:
Understand this has been "dismissed" as unnecessary in #133 but it doesn't negate the logic of why it's not included in the "Interface Options" menu.

XECDesign, 2022-01-06:
Duplicate of #133. `raspi-config` isn't intended to be a frontend for all the `config.txt` options.

closed 2022-01-06 by XECDesign