RPi-Distro/raspi-config #170: cmstart.sh: use xcompmgr -n in staed of xcompmgr -a

wvermin, 2021-11-17:
/usr/lib/raspi-config/cmstart.sh :

xcompmgr is started here with the flag '-a'. This affects at least two apps: xsnow and plank, resulting in (partially) black screens.

Would it be possible to use the flag '-n' in stead?

XECDesign, 2021-11-17:
Pretty sure `-a` is intentional.

@spl237 is xcompmgr still used on Bullseye or is it either mutter or openbox without xcompmgr?

spl237, 2021-11-17:
Bullseye uses either mutter or openbox with xcompmgr. And the -a flag is intentional - it fixes various rendering issues when used under openbox. The -n flag does composition in a completely different fashion which is not suitable for our needs.

If you have apps which require client-side rather than server-side compositing, you'll just need to change the flag in cmstart.sh yourself.

closed 2021-11-17 by spl237