RPi-Distro/raspi-config #171: FAKE KMS option removed ?

rnayabed, 2021-11-20:
Newest bullseye release no longer shows FAKE KMS as an option. It is far more stable and usable than the FULL KMS driver that is being shipped By default.

XECDesign, 2022-01-06:
Right, the focus is on improving the full KMS drivers. You can still use fake KMS by editing config.txt.

I believe the change is intentional.

closed 2022-01-06 by XECDesign

rnayabed, 2022-01-06:
> Right, the focus is on improving the full KMS drivers. You can still use fake KMS by editing config.txt.
> 
> I believe the change is intentional.

Instead of removing it from settings you should have at least first marked it with (Deprecated). A lot of screens are unusable with 3d accel as a result because many people aren't savvy enough to edit config and do it themselves.