RPi-Distro/raspi-config #173: dhcpcd dropin service file omits -q, leading to double logging

mutability, 2021-12-16:
The `wait.conf` dhcpcd dropin service file generated by raspi-config omits `-q` from the new command line.

(this was previously set by pi-gen when pi-gen generated the config itself rather than delegating to raspi-config - see RPi-Distro/pi-gen@ac246e300bdd55005ab811bcd817f1221b328fd7)

This leads to doubling up of dhcpcd logging (once direct to syslog, once via stdout/stderr)

XECDesign, 2022-01-06:
Previously there were complaints that we had `-q`, so I don't think either alternative will keep everybody happy here. IIRC, when updating things for bullseye I decided to bring our packages a little closer to Debian's, which omits `-q`.

Aside from preference, is there a reason you wouldn't want to see the messages in stdout?

mutability, 2022-01-06:
Well, dhcpcd appears to already log directly to syslog, so with the current config every message appears twice - once directly, once via systemd -> journald -> syslog

I'm not particularly attached to `-q` or not so long as the end solution doesn't log everything twice.

XECDesign, 2022-01-10:
Oh right, sorry. I thought you were saying that it was just that some messages would appear in stdout as well as the logs. I didn't notice that the logs were showing duplicate lines as a result. Yeah, that doesn't seem right at all. I've uploaded packages which should fix this.

mutability, 2022-01-24:
what's the more information that you need?

closed 2022-06-29 by XECDesign