RPi-Distro/raspi-config #174: Issues with VNC in headless mode

TheRaf974, 2021-12-22:
In somes cases, VNC display "cannot currently show the desktop" when I'm trying to connect to my Pi4 in headless mode.

For me, it's when I use the Pi camera in legacy mode (yes I can't use libcamera for the moment, I need raspicam).

The issue can probably be solved by two ways:

1) Just add the hdmi_force_hotplug=1, to force the rendering of the default desktop, as if the HDMI port was connected to a screen.

2) Instead of using vncserver-x11-serviced.service, we can use vncserver-virtuald.service. 
It creates a virtual desktop regardless to the HDMI.

In both cases, "VNC Resolution" became useless.

For the first solution, we have to set the default definition to have the results in VNC.

For the second solution,  the do_vnc_resolution have to modify the resolution of the virtual desktop but I don't know how.

XECDesign, 2021-12-23:
I'd go with option 1. You'd need to set the appropriate hdmi mode in config.txt to get a reasonable resolution.

I'm not sure if this is something raspi-config is meant to automatically help with. @spl237

TheRaf974, 2021-12-23:
That's what I have done.

By default, VNC is in service mode and we keep this mode by adding to config.txt these lines:
hdmi_force_hotplug=1
hdmi_ignore_edid=0xa5000080
hdmi_group=2
hdmi_mode=82

If we use a virtual desktop, it goes to virtual mode.

In fact, it's strange because, if the picam legacy mode is deactivated, there is no problem. I think there is a link with the GPU maybe.

With Raspberry Pi OS legacy, there is no problem.

spl237, 2021-12-23:
I think legacy camera mode on Bullseye combined with VNC is going to be something of a corner case; it's a bit of a hotch-potch of settings, because I suspect you don't have tvservice any more. I think you'll need to set resolution manually as you have done here.

TheRaf974, 2021-12-23:
Ok

I just realized that for the moment, I need legacy camera so I'm going to stay on buster and when libcamera will be more developped, I will go to bullseye

Thanks guys

closed 2021-12-23 by TheRaf974