RPi-Distro/raspi-config #175: [FR] Add option to use zram device for overlay upper

henryptung, 2022-01-05:
This should be a mostly-drop-in replacement for tmpfs in the overlay script, creating a journalless ext4 partition (mounted with `-o discard`) on top of a zram device instead of the current tmpfs.

XECDesign, 2022-01-06:
Would you happen to have any comparisons for boot times, CPU and RAM usage while doing things like installing large packages and using the browser?

henryptung, 2022-06-25:
In retrospect, probably not a great idea. tmpfs already supports swapping, so a zram swap device + normal tmpfs essentially allows for uncompressed "caching" while still leveraging zram as needed (vs. direct zram, which would force compression on all reads/writes).

See https://serverfault.com/a/843816/340187

closed 2022-06-25 by henryptung