RPi-Distro/raspi-config #178: When wireless network added, wpa_cli list_networks does not see it?

Jibun-no-Kage, 2022-01-15:
When wireless network added, wpa_cli list_networks does not see it?  If I use raspi-config nonint <ssid> <passpharse>, the /etc/wpa_supplicant/wpa_supplicant.conf has said network, but if I use wpa_cli list_networks, the list is empty.  This means that if wpa_cli is used to add any networks, it will step on the previous configuration.

XECDesign, 2022-01-17:
Are you passing the interface to `wpa_cli`? (For example, `wpa_cli -i wlan0 list_networks`)

Jibun-no-Kage, 2022-01-17:
Typically you don't have to do that, it finds the interface by default.  If I do add_network, etc. , via wpi_cli directly, I don't need to provide any interface.  Nor to the various examples showing use of list_networks quality the need for a specific interface.

For example...
wpa_cli list_networks
Selected interface 'p2p-dev-wlan0'
network id / ssid / bssid / flags
0       DD5_optout_nomap any
1       DD_optout_nomap  any

No explicit interface required.  Nor is any interface qualified in the actual conf file.  So I don't see the need per your question.

Using remove_network 0, or 1, then save_config also work with an explicit interface.

XECDesign, 2022-01-17:
I don't know why they changed that behaviour, but I find always specifying the interface, just works in all cases. There's not much I can do about it, so if you think it should do something different, it might be something to raise upstream.

closed 2022-01-17 by XECDesign

Jibun-no-Kage, 2022-01-17:
Yeah, I will change my ansible script to use the explicit interface, like raspi-config does.  Solves my issue.  But I think I will also asked why the behavior is such, it does 'feel' like the explicit interface and discovered interface are the same interface... from my specific.  A bit of catch 22 to be sure.