RPi-Distro/raspi-config #182: Command line option to enable and disable OverlayFS

mangelozzi, 2022-02-24:
To automate updates, its necessary to be able to be able to disable and enable overflayFS from the command line. I read many posts of people struggling with it, but I am not winning with hacking this script myself. I think many people would benefit from command line flags to be able to do this.
My mind is boggled looking at this script, absolutely amazing was done with bash, here I am riding the struggle bus on an in then else in bash.

Have you conisdered python?

bensonrodney, 2022-02-24:
Can't you already do this?
`sudo raspi-config nonint do_overlayfs 0`
(0 to enable, 1 to disable)

XECDesign, 2022-03-01:
raspi-config is intended to be a TUI not a CLI application. `nonint` exists so that our other applications can call functions from raspi-config. Then if there's an issue, we can fix it in one place rather than across multiple packages. It's not documented because it's not something that the average user is expected to use.

closed 2022-03-01 by XECDesign