RPi-Distro/raspi-config #187: non-interactive do_overlayfs also makes /boot read only

createcandle, 2022-03-21:
I'm trying to give a bash script the ability to enable the overlay feature using:

`raspi-config nonint do_overlayfs 0`

However it seems to set the /boot directory to read only too, which I don't want. 

The interactive mode asks if this is desired. What is the best way to control this behaviour in non-interactive mode?

XECDesign, 2022-03-21:
You can use `raspi-config nonint enable_overlayfs`, `raspi-config nonint enable_bootro`, `raspi-config nonint disable_bootro` and appropriate.

createcandle, 2022-03-21:
Thank you!

So if I understand correctly, doing this should give me the desired result?

```
raspi-config nonint enable_overlayfs
raspi-config nonint disable_bootro
```

*goes off to test

XECDesign, 2022-03-21:
Might want to disable_bootro first. I think enable_overlayfs temporarily makes boot writeable when necessary, but I didn't look too closely as it.

There's currently a bug in how ro detection is handled, which should be fixed with the next release. It may think boot is rw when it isn't, so when you try to toggle it it may say "nuh, we're already rw, I don't need to do anything".

createcandle, 2022-03-21:
Ok, so then I'll try

```
raspi-config nonint disable_bootro
raspi-config nonint enable_overlayfs
raspi-config nonint disable_bootro
```

createcandle, 2022-03-21:
Are you sure? This is my understanding:

1. disable_bootro to make sure `enable_overlayfs` will work ok
2. enable_overlayfs. This ends up making /boot read-only, which I don't want to happen.
3. disable_bootro again. To counteract number enable_overlayfs.

XECDesign, 2022-03-21:
`enable_overlayfs` should be independent of `enable/disable_bootro`. It will temporarily disable it to update the files it needs to update, but then re-enable it again, if that was the starting state.

createcandle, 2022-03-21:
Ah, I see. Thanks.

closed 2022-03-23 by createcandle