RPi-Distro/raspi-config #199: enabling and disabling wayland breaks lightdm autologin

pawalls, 2022-08-07:
**reproduction steps:**

1) enable wayland through raspi-config
2) disable wayland through raspi-config
3) enable autologin (GUI) through raspi-config
4) reboot or `/etc/init.d/lightdm restart`

**what happens:**

presented with login prompt.

`/var/log/lightdm.log` contains an error about `LXDE-pi-x` not being a valid session. i've confirmed there is no such session defined in the usual places. eg. `/usr/share/xsessions/`

**workaround:**

manually comment out the `autologin-session` and `user-session` values for Seat0 in `/etc/lightdm/lightdm.conf`

spl237, 2022-08-09:
I'm unable to repeat this with a fully-updated image; this was a problem with the initial Wayland release, but recent changes to session configuration should have fixed it. Please try again with all updates applied to a clean image and check the problem does not recur.

mikelpr, 2023-09-05:
@spl237 it is still a thing as of the 5th of September 2023, all updated on the latest raspberry os based on bullseye and also after dist-upgrading changing to bookworm. 
it seems that the user-session=LXDE-pi-wayland that lightdm.conf references just doesn't exist at all on the system. tried `find / -name "LXDE-pi-wayland*"` found nothing

spl237, 2023-09-05:
This all works on the forthcoming bookworm release; wayland was only ever an experimental feature on bullseye.

closed 2023-09-05 by spl237

mikelpr, 2023-09-05:
@spl237 I'm on bookworm and it's not working, am I missing a package?

spl237, 2023-09-05:
Unless you are in our beta programme, you are only able to access Debian's upstream bookworm packages and do not have access to our internal repo for our own bookworm packages; so you are missing multiple packages. This repo will only be made public when we officially release bookworm for Pi.

mikelpr, 2023-09-05:
@spl237 ahh gotcha it is a bit unfortunate tho that it is private. how can I get into the program? did not find any way googling and I'd like to test out an upgrade which I'd like to use wayland and I could give feedback on it

spl237, 2023-09-05:
I'm afraid the beta programme is by invitation only.

Our bookworm release should be ready within the next few weeks, barring any last-minute snags.