RPi-Distro/raspi-config #200: Please support a more flexible partition layout in the expand file system feature.

JinShil, 2022-08-25:
https://github.com/RPi-Distro/raspi-config/blob/2422cdc03548c219c556b71a1c3bb24c51b0b8b8/raspi-config#L181-L186

We need to add a data partition between the boot partition and rootfs partition.  This makes a total of 3 partitions and breaks the expand file system feature because it is hard-coded to only allow 2 partitions.

I suggest removing this above check as there is already a check that the partition to expand is the last partition at https://github.com/RPi-Distro/raspi-config/blob/2422cdc03548c219c556b71a1c3bb24c51b0b8b8/raspi-config#L188-L192

XECDesign, 2022-08-26:
This option is mostly deprecated by the automatic first boot resize. I can modify that to make it expand the root partition into whatever free space there is and remove checks that would prevent it from working.

What's the use case where you still have to run the resize through raspi-config? The only case I can think of is backup images being copied onto larger cards.

JinShil, 2022-08-29:
> What's the use case where you still have to run the resize through raspi-config? The only case I can think of is backup images being copied onto larger cards.

We don't have a specific use case to expand the rootfs partition through raspi-config.  In fact, we usually expand the root fs partition automatically on the first boot after loading our OS, just like the generic Raspberry Pi OS does.  But, we want our system to be compatible with the Raspberry Pi Foundation's software, so if a user needs to do a resize through raspi-config, for whatever reason, they can do that.

In this case, I don't see the check in question adding any value.  If the check were removed, and as long as the rootfs partition is last on the disk, I think the software should be able to expand the rootfs partition successfully.  This would allow users to have more flexibility in how they choose to partition their disk, but still remain compatible with the Raspberry Pi Foundation's software.

commit 82408f53 (2022-11-10): do_expand_rootfs: remove partition number check

Fixes #200
closed 2022-11-10 by XECDesign