RPi-Distro/raspi-config #202: Use of wpa_cli leaves in raspi-config wpa_supplicant file world readable

vtcifer, 2022-10-22:
The wpa_cli wrapper that raspi-config uses uses to setup access to wireless networks leaves `/etc/wpa_supplicant/wpa_supplicant.conf` file world readable:

```
root@raspberrypi:/etc/wpa_supplicant# ls -al
total 52
drwxr-xr-x  2 root root  4096 Oct 22 18:27 .
drwxr-xr-x 82 root root  4096 Sep 21 23:17 ..
-rwxr-xr-x  1 root root   937 Feb 25  2021 action_wpa.sh
-rw-r--r--  1 root root 25569 Feb 25  2021 functions.sh
-rwxr-xr-x  1 root root  4696 Feb 25  2021 ifupdown.sh
-rw-------  1 root root    73 Oct 22 18:27 wpa_supplicant.conf
root@raspberrypi:/etc/wpa_supplicant# raspi-config nonint do_wifi_ssid_passphrase  "Foo" "Bar" "0" "1"
root@raspberrypi:/etc/wpa_supplicant# ls -al
total 52
drwxr-xr-x  2 root root  4096 Oct 22 18:28 .
drwxr-xr-x 82 root root  4096 Sep 21 23:17 ..
-rwxr-xr-x  1 root root   937 Feb 25  2021 action_wpa.sh
-rw-r--r--  1 root root 25569 Feb 25  2021 functions.sh
-rwxr-xr-x  1 root root  4696 Feb 25  2021 ifupdown.sh
-rw-r--r--  1 root root    72 Oct 22 18:28 wpa_supplicant.conf
root@raspberrypi:/etc/wpa_supplicant#
```

While fixing wpa_cli to not leave this file with passwords in it world readable is certainly out of scope for here, addressing the fact that it happens isn't.  Fixing permissions to 600 after executing is likely needed here (as is done in other places when raspi-confg / wpa_cli isn't used, for example the rpi-imager has code that deals specifically with changing the permissions on this file, see link below).

https://github.com/raspberrypi/rpi-imager/blob/9d4665dbcac67d54c7c7e62690979580ce4c7349/src/OptionsPopup.qml#L723-L732

XECDesign, 2022-10-26:
Thanks for the report.

Using chmod would still allow the config file to be read for a brief moment, so that might be a bit less than ideal.

I've updated dhcpcd's wpa_supplicant hook to run it with a umask of 0077, which makes sure that wpa_supplicant.conf is always saved with 600. This update should be public once I've done a bit more testing to make sure it doesn't cause other issues.

XECDesign, 2022-10-31:
Updated version of dhcpcd has gone up.