RPi-Distro/raspi-config #212: Kernel update breaks booting when OverlayFS root is enabled

K-os, 2023-04-06:
I enabled the Overlay File System using raspi-config on my Bullseye Raspi OS system. Yesterday, I temporarily removed the parameter `boot=overlay` from /boot/cmdline.txt and rebooted the system to install some new packages and updates. With the updates a new kernel was installed (1.20230106-1 -> 1.20230405-1). After re-enabling `boot=overlay`, booting failed, because the initrd was not updated with the Linux kernel. Loading of the overlay module failed and subsequently, mounting the overlay root also failed.

I had to manually run `update-initramfs` and set the new initrd in `/boot/config.txt` to fix the problem. This should really be done automatically, by adding a hook in `/etc/kernel/postinst.d/` when enabling the OverlayFS using raspi-config.

I know, I could have disabled the overlay using raspi-config and re-enabled it afterwards using the same, tool. But for robustness it would be good for the initrd to be updated automatically.

OrangeDog, 2023-10-17:
The Overlay File System uses a completely different tool now, so may work better,

However, `update-initramfs` still writes to the wrong place with the wrong filenames.

XECDesign, 2023-10-17:
Works out of the box in bookworm.

closed 2023-10-17 by XECDesign

OrangeDog, 2023-10-17:
@XECDesign no it doesn't. If you do anything that updates the initramfs the new images created are e.g. `/boot/initrd.img-6.1.0-rpi4-rpi-v7l` instead of overwriting `/boot/firmware/initramfs7l`.

XECDesign, 2023-10-17:
Yeah, that's how it's meant to work. For general discussion about how and where the file actually get copied across, I'd recommend the forum.