RPi-Distro/raspi-config #213: do_update with nonint enters interactive mode

m5p3nc3r, 2023-05-03:
When trying to script usage of raspi-config, I wanted to ensure that the raspi-config tool was up to date.  The problem is that issuing the command

```
raspi-conifg nonint do_update
```
goes through the process of relaunching raspi-config into interactive mode.

The expected behaviour should be to return to the console prompt.

XECDesign, 2023-05-03:
nonint is only intended for use by other official scripts. I don't think we use `do_update` in nonint mode anywhere, but if you'd like to send a PR, I don't see why we couldn't merge it.

m5p3nc3r, 2023-05-03:
Understood - I have created PR #214

SpencerGraffunder, 2023-10-21:
It seems in doing some updates for Pi 5, the script has also removed the nonint feature from do_serial() (there used to be a nonint check after line 1186). It would be nice for that to be added back!

lurch, 2024-06-01:
#214 has been merged, so I guess this issue can be closed?

@SpencerGraffunder see my comment in #224

SpencerGraffunder, 2024-06-04:
I found the workaround of using do_serial_hw and do_serial_cons separately, so as far as I’m concerned it can be closed. Seems like it shouldn’t have changed for backwards compatibility though.

lurch, 2024-06-05:
> Seems like it shouldn’t have changed for backwards compatibility though.

I don't think that `raspi-config` has ever claimed to have a stable API? IIRC the non-interactive mode has always been "use this at your own risk" :wink: 

Perhaps that should be made more clear at https://www.raspberrypi.com/documentation/computers/configuration.html#raspi-config-cli @nathan-contino ?

closed 2024-06-12 by XECDesign