RPi-Distro/raspi-config #216: Manual page

da4089, 2023-07-24:
I went looking for a manual page for raspi-conf and couldn't find one.

Would the maintainer be interested in merging one if it was available?

XECDesign, 2023-07-24:
Maintaining it would be the problem. I'm guessing that you were after documentation for non-interactive usage?

That area is undocumented somewhat intentionally. It may change frequently without notice, so it might be a bad idea for others to build on it.

da4089, 2023-07-24:
Yes, the nonint commands were my primary interest.

The Raspberry Pi Foundation web site has fairly detailed information (https://www.raspberrypi.com/documentation/computers/configuration.html#list-of-options26) so ... it seems maybe that keeping it as a "private" undocumented API isn't really possible now?

I'd be happy to continue to maintain the man page, derived from a combination of the official web page and analysis of the Bash code.

XECDesign, 2023-07-25:
@spl237 thoughts?

spl237, 2023-07-25:
Many thanks for the offer, but if there is to be one, then it really ought to be written and maintained by our docs team rather than a third party.

But in general, people really shouldn't be using the nonint commands - they are all designed for use by the Raspberry Pi Configuration GUI app, and as @XECDesign says, they may well change underneath people.

da4089, 2023-07-25:
Ok.  Thanks for considering it.

closed 2023-07-25 by da4089