RPi-Distro/raspi-config #219: Upgraded to bookworm (Raspbian 12), "Overlay File System" now fails to enable

ElTopo, 2023-08-15:
I use raspi-config to enable/disable "Overlay File System".

raspi-config worked well on bullseye (Raspbian 11).

After I upgraded to bookworm (Raspbian 12), I can't enable Overlay file system.

With raspi-config GUI, when I tried to enable overlay file system, raspi-config fails with the screen: 

**The boot partition is currently writable. This cannot be
 changed while an overlay file system is enabled.**

When I tried command
`sudo raspi-config nonint do_overlayfs 0`
I got the following errors:
**update-initramfs: Generating /boot/initrd.img-6.1.21+
grep: /boot/config-6.1.21+: No such file or directory
W: zstd compression (CONFIG_RD_ZSTD) not supported by kernel, using gzip
grep: /boot/config-6.1.21+: No such file or directory
E: gzip compression (CONFIG_RD_GZIP) not supported by kernel
update-initramfs: failed for /boot/initrd.img-6.1.21+ with 1.
mv: cannot stat '/boot/initrd.img-6.1.21+': No such file or directory**

raspi-config version: 20230214 
Raspbian: bookworm (12) 
kernel version: Linux rpi 6.1.21+ #1642 Mon Apr  3 17:19:14 BST 2023 armv6l GNU/Linux

ElTopo, 2023-08-16:
Further inspection found that this was not a bug from raspi-config, but one of /usr/bin/unmkinitramfs from package "initramfs-tools-core/testing,now 0.142". 
Reverted to "initramfs-tools-core/stable,now 0.140", and raspi-config works again.
Case closed.

frantisekstanko, 2024-11-24:
The latest Raspbian, released on 2024-11-19, still seems to be affected. Interestingly, the overlay filesystem works properly on Raspberry PI 3 and 4, but not on 5.

ElTopo, 2024-11-26:
@frantisekstanko in 2023, Raspbian/RaspberryPiOS made a major change of kernel packages.
Old systems use raspberrypi-kernel, which is a huge package and has all kernels for for all RPI models, it uses **/boot** partition.
New systems use linux-image-*, which are smaller packages for individual RPI models, it uses **/boot/firmware** partition. 
The later releases of raspi-config just discarded old systems and only works for new systems.

If your RPI is still running kernel 6.1.21 with /boot, you'll need to use an old release of raspi-config, or download a new Raspberry Pi OS image to rebuild your SD card. The old system's latest kernel is raspberrypi-kernel 1:1.20230405-1 and probably won't another update in the future.

frantisekstanko, 2024-11-26:
> @frantisekstanko in 2023, Raspbian/RaspberryPiOS made a major change of kernel packages. Old systems use raspberrypi-kernel, which is a huge package and has all kernels for for all RPI models, it [...]

That's an interesting piece of information and now I understand why the move to `/boot/firmware`. However, I fail to understand the solution to the problem that you propose, even though I have re-read your comment multiple times.

I have flashed SD cards with the `2024-11-19` image (released a week ago as of this writing). It is running kernel `6.6.51+rpt-rpi-v8`. With this image and a freshly flashed SD card, the overlay works properly on Raspberry Pi 3 and/or 4 devices, but not on 5.

Am I misunderstanding something and did you propose a solution to this problem?

ElTopo, 2024-11-26:
@frantisekstanko I don't have a RPI 5, but after I re-imaged my RPI 3B+ raspi-config's overlay rootfs works the the new system, I would assume it should also work with RPI 5 with new system.

This issue was regarding the problem when the official Raspbian repo had the **old** raspi-config package with the new initramfs-tools-core package. I believe the repo had been updated after.