RPi-Distro/raspi-config #229: Enabling VNC for wayland desktop does nothing since wayvnc unit file does not exist

andrum993, 2023-12-15:
raspi-config was recently updated to use a systemd service to run the wayvnc VNC server, instead of the previous method. Unfortunately when raspi-config attempts to enable or disable wayvnc it now does nothing, since the referenced wayvnc unit file does not exist. I'm guessing that there is an updated wayvnc package in the works, but in the meantime raspi-config's option to enable and disable VNC is non-functional on systems where the desktop is using wayland, which is all Pi 4, Pi 400, CM4 and Pi 5 devices by default (but only on those devices which use the graphical desktop, of course).

Also, raspi-config itself appears to not handle the failure to enable the VNC server properly:
[image]

Tested on a Pi 4.

andrum993, 2023-12-15:
See also https://github.com/raspberrypi/bookworm-feedback/issues/174

XECDesign, 2023-12-18:
Fixed in 77c6b95d3bea51125a9c523164edc336d0b6c53c

closed 2023-12-18 by XECDesign