RPi-Distro/raspi-config #231: No "Audio" menu with raspi-config 20231216 in Zero W

toobaz, 2023-12-17:
I am trying to follow [a guide](https://learn.adafruit.com/adding-basic-audio-ouput-to-raspberry-pi-zero/pi-zero-pwm-audio) to get audio from the PWM pins of my Raspeberry Pi Zero W. 

I correctly applied the dt overlay - as confirmed by the following commands:

```
pietro@pyplay:~ $ pinctrl get GPIO18
18: a5    -- | lo // GPIO18 = PWM0
pietro@pyplay:~ $ pinctrl get GPIO13
13: a0    -- | lo // GPIO13 = PWM1
```

Next step would be to use ``raspi-config`` to force the use of the 3.5 jack.

However, when I run ``raspi-config`` and select ``System Options`` and then ``S2 Audio``, the interface simply goes back to the main menu. No audio menu is shown.

This bug is likely related to https://github.com/RPi-Distro/raspi-config/issues/147 , but the symptoms are different.

Ultimately, I found out the problem is solved by giving 

```
pietro@pyplay:~ $ sudo rmmod snd_bcm2835
pietro@pyplay:~ $ sudo modprobe snd_bcm2835 enable_headphones
```

... but I guess ``raspi-config`` should show a message "no audio interface found"  (and possibly suggest the command above, or some source of documentation) rather than silently coming back to main menu.

j-davis-2022, 2024-07-13:
I can reproduce this on version 20240708.

XECDesign, 2024-07-16:
Since the vast majority of Pi Zero users won't have an audio jack to switch to, I think showing the option is more likely to cause issues and confuse people.

This seems a bit outside the scope of what raspi-config is for and is more of a documentation issue.

toobaz, 2024-07-16:
> I think showing the option is more likely to cause issues and confuse people.

Then please hide "S2 Audio"... clicking on a menu item that does nothing (not even printing "no audio available"), that definitely confuses people.

XECDesign, 2024-07-16:
Fair point. And upon further inspection is looks like it's not behaving the way it's meant to at all.

toobaz, 2024-07-16:
> Fair point. And upon further inspection is looks like it's not behaving the way it's meant to at all.

Thanks! And I forgot to mention that this looks like a regression - the guide I referred to initially seems to imply that the submenu used to appear back in December 2015.