RPi-Distro/raspi-config #232: feat: print out config values in nonint mode

guerda, 2024-01-03:
It's great that `raspi-config` lets you set values in an non-interactive way via `nonint`.

In order to make this even better, it would be great if you also can read out the existing configuration via non-int mode, to backup configuration etc.

It could look like this:

    $ sudo raspi-config nonint do_audio
    0

I don't know if that's possible for all options, but it could be a good start to let you automate the config.

XECDesign, 2024-01-04:
It's a bit outside of the scope of what raspi-config is meant to be for. The nonint functionality is primarily there for rc_gui, not general use. The interface can and will change without any notice.

There are some get_* functions there already, but again, this isn't something we're claiming to provide support for.

guerda, 2024-01-04:
OK, I understand that. Maybe that should be put as a warning notice on the documentation page, too.

XECDesign, 2024-01-04:
Yup, I think that can be done as an issue or a PR [here](https://github.com/raspberrypi/documentation/).

closed 2024-01-04 by XECDesign