RPi-Distro/raspi-config #235: A5 - Bootloader version both options returns a return to default

GrayHatGuy, 2024-01-17:
When selecting options for bootloader version both 
E1 - Latest Use latest bootloader version 
and 
E2 - Default Use the factory default bootloader 
Return with message that default bootloader version has been installed to eeprom
So E1 does not update to latest and both E1 and E2 return the bootloader to default

GrayHatGuy, 2024-01-17:
I verified this by flashing a Bootloader update to SD card via Raspberry Imager and booting to update eeprom with successfully booted from external USB then I tried updating the latest bootloader from Raspi-config using E1 - Latest to update and the eeprom and it reverted to factory and refused to boot to external USB.

XECDesign, 2024-01-18:
@timg236

timg236, 2024-01-18:
I can't see any difference in behaviour. Selecting Latest/Default uses the embedded EEPROM config from the file in lib/firmware which means the BOOT_ORDER will be blank and therefore default to 0xf41 which is the default SD followed by USB.

GrayHatGuy, 2024-01-18:
> I can't see any difference in behaviour. Selecting Latest/Default uses the embedded EEPROM config from the file in lib/firmware which means the BOOT_ORDER will be blank and therefore default to 0xf41 which is the default SD followed by USB.

I could be experiencing other prevailing issues at minimum the menu text return after choosing either seems to indicate default eeproms were updated for both cases

GrayHatGuy, 2024-01-22:
I figured out my problem I had a bad usb to IDE/SATA cable causing intermittent disk rw issues. It does not appear to be eeprom confirmed by using USB thumb drives