RPi-Distro/raspi-config #237: Confusing messages when switching Bootloader version

peterharperuk, 2024-02-02:
raspi-config has an option in the advanced menu to switch the bootloader between the latest and default.
If you switch to latest a message comes up saying "Reset boot ROM to defaults?"
The word "defaults" is confusing as I've asked it to switch to "latest".
If I choose "yes" it says "Boot ROM reset to defaults" - again confusing, as I've switched to the latest image.
If I choose "no" it says "Boot ROM not reset to defaults" - which is true, because the No button seems to have not effect.

I think we should refer to Bootloader rather than Boot ROM and stop referring to "default" when switching to latest.

timg236, 2024-02-02:
I'm happy to drop "ROM" - it will revert the EEPROM config to default settings though

peterharperuk, 2024-02-02:
I think it's the word "default" I find confusing. And I'm supposed to know what I'm doing :)

timg236, 2024-02-02:
"Default release" or "Factory default release and settings" might be clearer.

Of course the factory default changes over time so it might be newer than what you are on now but I think most people understand the concept of "Factory default"

mobluse, 2024-02-03:
I think "Reset boot ROM to defaults? Yes" means that it after installation will be set to the defaults for the new ROM that has just been installed. Is this correct?