RPi-Distro/raspi-config #244: Suggestion: Delete and restore unused kernels and initrds

framps, 2024-05-24:
This is no issue but a suggestion for an additional feature of raspi-config.

I run an old Raspberry 2B with bookworm and every time I update the system a lot of kernels and initrds are rebuild even the system only uses one kernel and initrd. I wrote [this script](https://github.com/framps/raspberryTools/blob/master/raspiHandleKernels.sh) for me which deletes all unused kernels and initrds and also allows to install them again them if the system should run on another Raspberry HW. Then OS updates are completed much faster.

Wouldn't this be a useful new feature of raspi-config? I have no clue how many old Raspberries are still used out there and whether from this perspective it makes sense to add this feature. 

JM2C :wink:

XECDesign, 2024-05-24:
Yup, I was thinking we'll need something like that when I was switching to Debian's packaging for the kernel.

I'd probably lean towards removing the apt packages for unused kernels rather than just the files in /boot. But we've got to be careful with the situation where someone has updated, but not yet rebooted. We don't want to remove modules which a user may still want to load and we don't want to remove a kernel which we would be booting into on next reboot.

Would you like to send a PR or is this just a suggestion?

framps, 2024-05-24:
> I'd probably lean towards removing the apt packages for unused kernels rather than just the files in /boot.

Looks like  you didn't check my code :smiley: I basically check which kernels are installed and which kernel is used right now. Then I create a file with a list of the kernels which will be removed so they can be reinstalled later on and remove the kernel packages from the system with apt :wink: 

> Would you like to send a PR or is this just a suggestion?

Well, it's a suggestion as of now. I didn't know you'll accept a PR from me. There are 10 open PRs and I don't want to spend effort in a PR which stays there forever :wink: 

If you check my code in my [little script](https://github.com/framps/raspberryTools/blob/master/raspiHandleKernels.sh) I wrote and confirm that's a valid way to cleanup and restore the kernels I will check rpi-config and will create a PR. As far as I can see option 4 (performance option) will be the right place to add this new feature.

XECDesign, 2024-05-24:
If you're worried that the PR may be left hanging for a long time, then it's best to hold off. I'm away for a bit and can't promise I won't we swamped with higher priority tasks when I get back.

When I get the time, I can take inspiration from your script and add something similar to raspi-config.

framps, 2024-05-24:
Ok. So in general you're willing to accept a PR from me. That's great and was my major concern. 

I just created a fork and checked rpi-config. As far as I can see it's not much effort to add another function which uses my logic to handle the kernels and to add this function in the rpi-config menu hierarchy. I'll create a PR and crossing fingers you're not swamped with higher prio tasks when you're back :wink:

framps, 2024-08-23:
> I'd probably lean towards removing the apt packages for unused kernels rather than just the files in /boot. 

I just synced my PR and noticed you asked to remove the apt package. That's what I do in my PR :wink:

XECDesign, 2024-08-23:
Thanks! And I haven't forgotten about this, sorry it's taking a while.

framps, 2024-08-24:
No hurry.

I just want to update you the PR uses apt to remove the unused kernels - that matches  your preferred way to handle this issue - and the files in /boot are not just deleted :wink:

framps, 2024-08-24:
I just noticed I missed this kernel cleanup should only be available starting with Bookworm (debian >=12) . Not sure right now how to handle this in in the rpi-clone menu :-(

framps, 2024-08-25:
Just detected raspi-config will report "No unused kernels found" on Bullseye and previous OS releases and there is no check for Bookworm required.

framps, 2024-08-25:
> I'd probably lean towards removing the apt packages for unused kernels rather than just the files in /boot.

I updated the PR to use dpkg to retrieve the installed kernels and not use the kernel source files in /boot any more. I think that's what you had in your mind and is a much better solution :+1:

framps, 2025-01-08:
@XECDesign I spent some time to create [this PR](https://github.com/RPi-Distro/raspi-config/pull/245) which is now pending for 8 months. Would be great to get any feedback and why it's not merged :thinking:

XECDesign, 2025-01-09:
Alright, I'll be sure to take a look this weekend. Thanks for the reminder.