RPi-Distro/raspi-config #253: Add support for newer Lua versions

mimi89999, 2024-08-28:
Hello,
Currently `raspi-config` depends on `lua5.1` (https://github.com/RPi-Distro/raspi-config/blob/bookworm/debian/control#L12). Would it be possible to add support for newer versions of Lua and not depend on that very old Lua?

mimi89999, 2024-08-29:
I tested the code with Lua3.4 and it seems to be working fine. I fear that switching to sed might potentially introduce new bugs and make the code less readable.

mimi89999, 2024-08-29:
Anyway, the `lua` command is called and it's handled by the Debian alternatives system:
```
michel@kokomi:~ $ which lua
/usr/bin/lua
michel@kokomi:~ $ ls -l /usr/bin/lua
lrwxrwxrwx 1 root root 33 Feb 21  2023 /usr/bin/lua -> /etc/alternatives/lua-interpreter
michel@kokomi:~ $ ls -l /etc/alternatives/lua-interpreter
lrwxrwxrwx 1 root root 15 Aug 28 21:09 /etc/alternatives/lua-interpreter -> /usr/bin/lua5.4
michel@kokomi:~ $ sudo update-alternatives --list lua-interpreter
/usr/bin/lua5.1
/usr/bin/lua5.4
```

So even though the package depends on `lua5.1`, the `lua` command could be any version of lua :man_shrugging:

XECDesign, 2024-08-29:
Thanks for checking and that's a good point about the lua symlink. But I still think the lua snippets are a bit out of place in a shell script. I don't think it will be any less readable. It would be using the same regular expressions, which is usually the least readable part.

closed 2025-10-31 by XECDesign