RPi-Distro/raspi-config #266: Overlay Filestsystem doesn’t seem to work anymore

vasilisvg, 2025-01-12:
Since a few days the overlay file system doesn’t work when I install it via raspi-config. After the reboot, the default file system loads, and changes I make are persistent. 

````
Linux pi-paper-1 6.6.51+rpt-rpi-v7 #1 SMP Raspbian 1:6.6.51-1+rpt3 (2024-10-08) armv7l
````

This is the 32bit version, headless, on a raspberry pi 3a+. I’ve had the same issue on another 3a+. I haven’t tested another pi yet.

SosenWiosen, 2025-01-22:
Hi, I also get this issue on the Raspberry PI CM 5. Clean install of bookworm, first thing I do is to set overlay and read only memory, and then reboot. All the changes, new files etc. are persistent :(

wdormann, 2025-02-05:
Can confirm with 32-bit Lite Raspbian bookworm (`Linux pizerow 6.6.51+rpt-rpi-v6 #1 Raspbian 1:6.6.51-1+rpt3 (2024-10-08) armv6l GNU/Linux`).  After enabling the overlay filesystem with `raspi-config` and rebooting, dmesg shows:
```
wd@pizerow:~ $ mount | grep overlay
wd@pizerow:~ $ dmesg | grep overlay
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_headphones=0 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_hdmi=0  smsc95xx.macaddr=B8:27:EB:56:CC:4E vc_mem.mem_base=0x1ec00000 vc_mem.mem_size=0x20000000  overlayroot=tmpfs console=ttyS0,115200 console=tty1 root=PARTUUID=bb804608-02 rootfstype=ext4 fsck.repair=yes rootwait cfg80211.ieee80211_regdom=US
[    0.000000] Unknown kernel command line parameters "overlayroot=tmpfs", will be passed to user space.
[    6.092306]     overlayroot=tmpfs
wd@pizerow:~ $ grep overlay /boot/firmware/cmdline.txt 
overlayroot=tmpfs console=serial0,115200 console=tty1 root=PARTUUID=bb804608-02 rootfstype=ext4 fsck.repair=yes rootwait cfg80211.ieee80211_regdom=US
wd@pizerow:~ $ 
```

`/etc/overlayroot.conf` has:
```
overlayroot_cfgdisk="disabled"
overlayroot=""
```

And the output of the `mount` command is completely unchanged between the before-enabling and after-enabling-and-rebooting.

Perhaps `raspi-config` is expecting something that's not present in the 32-bit and/or Lite version of Raspbian. But as it is, suggesting that one can enable the overlay filesystem on it is deceptive.

ogaclejapan, 2025-02-11:
Hi, I'm running into the same issue on my Raspberry Pi 5 when using `kernel8.img`.

Looking at the `overlayroot` logs, I see the following error:

```
$ sudo cat /run/initramfs/overlayroot.log
builtin set cfgdisk='disabled'
Unable to find driver/module. searched: overlay overlayfs
[failure]: Unable to find a driver. searched: overlay overlayfs
```

I tried adding the overlay module to initramfs, but it didn’t make any difference:

```
$ sudo cat /etc/initramfs-tools/modules
# List of modules that you want to include in your initramfs.
# They will be loaded at boot time in the order below.
#
# Syntax:  module_name [args ...]
#
# You must run update-initramfs(8) to effect this change.
#
# Examples:
#
# raid1
# sd_mod
overlay

$ sudo update-initramfs -u
$ sudo reboot
```

I'm stuck trying to get overlay working with this kernel8.img :(

Interestingly, it works fine with the default Raspberry Pi 5 kernel (which uses a 16KB page size), but I'm using kernel8.img because I need a 4KB page size, as some packages require it. The page size issue discusses that in more detail:
https://github.com/raspberrypi/bookworm-feedback/issues/107#issuecomment-1773810662

```
Linux rpi-demo 6.6.74+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.74-1+rpt1 (2025-01-27) aarch64 GNU/Linux
```

OlehAhieienko, 2025-03-15:
I have overlayfs enabled on Raspberry Pi 4 with Debian 12.9 and it works just fine.
Disabled it for Debian 12.10 and firmware update. After that, enabled it back via `raspi-config`. Options is showed as enabled in menu, but it is not enabled on system itself.

Mingendahl, 2025-04-20:
Same here. RPI4, bookworm.
Usually, when activating overlayFS, it usually does things (don't remember exactly what, but it works for some seconds)... now it just immediately tells me it's enabled and I need to reboot. But it is not enabled... Any idea?

Mingendahl, 2025-04-21:
Argh... checked it on a second RPi4... same issue. :/

Mingendahl, 2025-04-21:
Ok. Don't trust people who know a 'safe' way to migrate from Bullseye to Bookworm...
After a fresh install, everything works as expected.

XECDesign, 2025-05-02:
> Syslog shows
> Unknown kernel command line parameters "splash overlayroot=tmpfs", will be passed to user space.

That's normal.

I've just flashed the last [lite armhf image ](https://downloads.raspberrypi.com/raspios_lite_armhf/images/raspios_lite_armhf-2024-11-19/2024-11-19-raspios-bookworm-armhf-lite.img.xz), booted it on a 3B+, ran through the new user prompts, updated all the packages, enabled overlayfs through raspi-config, rebooted and it all worked as expected.

Same test passed on a 4B+ and a Zero W

There have been some issues in the past with some initramfs files not being updated, but that should be resolved now.

@ogaclejapan I think you're running into [this issue](https://github.com/raspberrypi/firmware/issues/1965). I was able to reproduce it on a pi5 with kernel=kernel8.img. There's a fix for that in the pipeline, but it hasn't shipped for the affected boards yet. As a workaround, you can add `initramfs initramfs8 followkenel` to your config.txt file.

If anybody is still having issues, please provide steps to reproduce them and make sure all your packages are updated before you start.

closed 2025-05-27 by XECDesign