RPi-Distro/raspi-config #282: feature request: activate / deactivate Wifi Power saving mode via raspi-config

hubertmeier, 2025-08-31:
Power saving for wifi is by default on. That causes often problems if you want to reach the raspi when the raspi is connected per wlan.

sure, there are workarounds like

sudo iw wlan0 set power_save off

but this has to be startet at every boot. It would be helpful to have this configurable in raspi-config

pelwell, 2025-09-02:
It was off for quite a while, but the desire to reduce our power consumption led to reverting that kernel patch a few years ago:
```
[    8.422754] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
```
`sudo iw wlan0 set power_save off` is not a workaround, it's the way Linux expects power saving to be configured. Put it in a dedicated systemd service, or create `/etc/rc.local` and run `sudo systemctl enable rc-local.service`.

XECDesign, 2025-09-05:
https://www.freedesktop.org/software/systemd/man/latest/systemd-rc-local-generator.html
```
Support for /etc/rc.local is provided for compatibility with specific System V systems only. However, it is strongly recommended to avoid using this script today, and instead provide proper unit files with appropriate dependencies for any scripts to run during the boot process. Note that the path to the script is set at compile time and varies between distributions.
```

Maybe a udev rule would be more fitting for this, but it also feels like something that should maybe be a module parameter?

XECDesign, 2025-09-05:
Makes sense.

Looks like it can be done through NM as well.
https://networkmanager.dev/docs/api/latest/settings-802-11-wireless.html

Haven't tried either, but Claude suggests these two alternatives to requiring a systemd service.
`/etc/NetworkManager/conf.d/default-wifi-powersave-off.conf`
```
[connection]
wifi.powersave = 2
```

Or a udev rule,
`/etc/udev/rules.d/70-wifi-powersave.rules`:
```
ACTION=="add", SUBSYSTEM=="net", KERNEL=="wlan*", RUN+="/usr/bin/iw %k set power_save off"
```

The reason I am steering away from system services altogether because making sure it runs at the right point in time can be a pain and you end up requiring getting udev involved anyway.

pelwell, 2025-09-05:
See https://github.com/raspberrypi/linux/pull/7038.